pazi88/STM32_mega #1: Arduino digital pins 22-53 doesn't work

KUSTOMSS, 2024-02-06:
Hi digital pins 22-53 doesn't work.
On arduino in my tune work everything.
Any solutions?

pazi88, 2024-02-07:
Are you using the pre-compiled binary?

KUSTOMSS, 2024-02-07:
Yes i tried:
- 202305_FastADC_InternalFlash.bin
- 202305_FastADC_InternalFlash_FTDI.bin
- 202305_InternalFlash.bin
- 202305_InternalFlash_FTDI.bin
- 202305_InternalFlash_FTDI.bin
- 202207_FastADC_InternalFlash.bin
-[202207_InternalFlash.bin

I ordered 2 boards with assembly from: 
https://www.pcbway.com/project/shareproject/STM32_mega_b8a000f3.html

KUSTOMSS, 2024-02-14:
I Found solution. It's not problem with 22-53 digital pins it's problem's with 22-23 digital pins.
PC9 - D22
PC8 - D23
This 2 pins is used in code to SD
  {PC_8,  SDIO, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF12_SDIO)}, // SDIO_D0
  {PC_9,  SDIO, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF12_SDIO)}, // SDIO_D1
  
 Can i disable this pin to SD and it will be work tu aux digital input?? In schematics it doesnt be used to SD

pazi88, 2024-02-20:
> I Found solution. It's not problem with 22-53 digital pins it's problem's with 22-23 digital pins. PC9 - D22 PC8 - D23 This 2 pins is used in code to SD {PC_8, SDIO, STM_PIN_DATA(STM_MODE_AF_PP, [...]

The pins 22 and 23 don't currently work properly, because those are mapped wrong in the variant file. On purpose though to make serial connection to TS working. The SDIO is not used with speeduino, so it doesn't matter.

pazi88, 2024-02-20:
> digital pin 46 doesn't work. digital pin 51,53,24 work when sd card is removed.

I can't think of any reason why pin 46 won't work. Unless you have WMI somehow enabled. It's default pin in there at least in 0.4 board. for the others, I can't think any explanation why those won't work.