pazi88/STM32_mega #6: Pin 20 vss

ooSPIDIIoo, 2024-10-12:
Hello, I'm checking the data and pin 20 of the Arduino Mega is the one dedicated to VSS. According to what I see in the schematics of the STM32 adapter board, the pin is not connected. Is it possible to use another one? I currently use the speed signal on my Speeduino.
Thanks

alexandrefelipemuller, 2024-10-14:
to use VSS on speeduino the pin must receive interrupts... but you have to change on code

pazi88, 2025-02-11:
> Hello, I'm checking the data and pin 20 of the Arduino Mega is the one dedicated to VSS. According to what I see in the schematics of the STM32 adapter board, the pin is not connected. Is it possible to use another one? I currently use the speed signal on my Speeduino. Thanks

There unfortunately isn't any free inputs on the F407 processor, so it's not as simple as just adding it. The current design has been done based on what Arduino Mega pins has been used on Speeduino boards, and I don't remember pin20 being used on any of those.